Annex-Engineering/klipper_estimator #2: FR: consider adding it to to metadata extract running at the moonraker file upload

zellneralex, 2022-01-14:
I know it might be to early to do it know but it would be great if that could be integrated in the metadata extract running at every gcode file upload within moonraker.

I am fully aware that this means maybe a change of coding language.

There is a other project (execute objects) that already included a post process via a pip package so a other solution could be to integrate these 2 together

dalegaard, 2022-01-14:
Hi @zellneralex 

I'm not sure I understand what you mean with this? Do you want to run `klipper_estimator` on every file uploaded via Moonraker?

If yes(which could work, though it might be slow because it's running on a Raspberry Pi) then that would be for the Moonraker project to do I think :-)

Best regards,
Lasse

zellneralex, 2022-01-20:
Hi Lasse,

sorry for my late response. What did you expect ;-) a day after you published it at the klipper discord there where the first requests to add it in the UIs to get a better prediction on ETA.
And therefor I thought I start here as I know you would be capable to do it and do not wont to overload Arksine.
But I can understand your answer.

I can close it if you want that.

Best regards
Alex

Arksine, 2022-01-22:
I'm going to look into adding support for this,  I started playing with it and I find it pretty exciting.  On a Pi 3 it's reasonably fast, I ran a 200MB file in under 45s.  When I wrote Moonraker's metadata parser it was important to me that it be quick, so it relies completely on what can be extracted from the comments (and it some cases from the gcode).  The Klipper Estimator is more akin to Octoprint's gcode processor, however its FAST.

dalegaard, 2022-01-22:
@Arksine that sounds great, let me know if you need any special output or find something missing. I recently added a `--format json` option to the `estimate` command so quite extensive output is available.

Piezoid, 2022-01-22:
When dalegaard was "thinking of adding some generic metadata parsing support to the planner library, so clients can easily get stuff like layer height", I suggested implementing moonraker's `metadata.py` json interface such that `klipper_estimator` can be used as a drop-in replacement for this script. `metadata.py` already runs in a separate process, thus it doesn't matters whether it use the python runtime or not.

Otoh, re-implementing a compatible and throughout metadata extraction might be out of scope for `klipper_estimator`, since `metadata.py` is almost ~1kLOC.

dalegaard, 2022-03-13:
I'll close this for now, feel free to post any updates here and we'll re-open if relevant :)

Best regards
Lasse

closed 2022-03-13 by dalegaard