Annex-Engineering/klipper_estimator #4: Error on post-process (estimate working fine)

jankolkmeier, 2022-01-17:
Whenever I run klipper_estimator in ```post-process``` mode, I get the following error:

```
>>> .\klipper_estimator.exe --config_moonraker_url http://ratos.local post-process SmallCube.gcode
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', tool/src/cmd/post_process.rs:289:47
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
I've tried a number of different g-code files, both with v1.0.0 and v1.1.0, and I've tried both on Windows (```klipper_estimator.exe```) and on my RPI (```klipper_estimator_rpi```). Running ```estimate``` returns the expected result (see below). 

- Slicer: SuperSlicer 2.3.57
- OS: Windows 10 / RatOS (a MainsailOS flavor)
- Klipper: v0.10.0-225-g6e6ad7b5
- Moonraker: v0.7.1-285-g505c1bf

[Sample of a gcode file](https://gist.github.com/jankolkmeier/305e9532e1aae1593bd8a0f558d76de8) that failed for me.
Result of this file using ```estimate```:
```
Sequences:
 Run 0:
  Total moves: 905
  Total distance: 2180.537mm
  Total extrude distance: 51.309mm
  Minimal time: 1m4.534s (64.534s)
  Average flow: 1.912368528150471 mm³/s
  Phases:
   Acceleration: 6.882s
   Cruise:       50.966s
   Deceleration: 6.686s
  Move kind distribution:
   43.971s              => Internal perimeter
   17.541s              => External perimeter
   1.068s               => Custom
   0.993s               => Firmware retract
   0.962s               => Firmware unretract
  Layer time distribution:
thread 'main' panicked at 'index out of bounds: the len is 16 but the index is 16', tool/src/cmd/estimate.rs:257:55
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
   0.000: 0.031s    0.320: 1.955s    0.640: 1.956s    0.960: 1.956s    1.280: 1.956s    1.600: 1.956s    1.920: 1.956s    2.240: 1.956s    2.560: 1.956s    2.880: 1.956s    3.200: 1.956s    3.520: 1.956s    3.840: 1.956s    4.160: 1.956s    4.480: 1.956s    4.800: 1.956s
```

jankolkmeier, 2022-01-17:
Not sure if relevant, but here is the result of ```dump_config```:

```json
{
  "max_velocity": 400.0,
  "max_acceleration": 5000.0,
  "max_accel_to_decel": 2500.0,
  "square_corner_velocity": 5.0,
  "instant_corner_velocity": 1.0,
  "move_checkers": [
    {
      "axis_limiter": {
        "axis": [
          0.0,
          0.0,
          1.0
        ],
        "max_velocity": 15.0,
        "max_accel": 120.0
      }
    },
    {
      "extruder_limiter": {
        "max_velocity": 60.0,
        "max_accel": 2078.758440383939
      }
    }
  ],
  "firmware_retraction": {
    "retract_length": 0.5,
    "unretract_extra_length": 0.0,
    "unretract_speed": 40.0,
    "retract_speed": 40.0
  }
}
```

dalegaard, 2022-01-17:
This is caused by `firmware_retraction` so yes the config was very helpful :blush:

dalegaard, 2022-01-17:
Thanks for reporting this. New version is going up with this fixed.

commit 1af9f5c6 (2022-01-17): Handle gcode expanding to multiple operations

Fixes #4
closed 2022-01-17 by dalegaard