Annex-Engineering/klipper_estimator #5: FR: Arc Support

samwiseg0, 2022-01-23:
When running `klipper_estimator` on an arc-welded file it provides wildly incorrect estimates. Ideally, the gcode file could be processed through `klipper_estimator` after its been arc-welded.

dalegaard, 2022-01-23:
Hi!

Currently gcode arcs aren't supported. I'll look into how difficult it would be to support.

Best regards,
Lasse

SnorreSelmer, 2022-06-15:
Klipper recommends _against_ using arcs... You save a bit of file-size, but gain nothing in quality over not using arcs.

https://klipper.discourse.group/t/arc-movement-g2-g3/490/3

samwiseg0, 2022-06-15:
> Klipper recommends _against_ using arcs... You save a bit of file-size, but gain nothing in quality over not using arcs.
> 
> [klipper.discourse.group/t/arc-movement-g2-g3/490/3](https://klipper.discourse.group/t/arc-movement-g2-g3/490/3)

Not seeing where it is not recommended? Regardless, asking to support something that klipper itself supports is valid and whether or not it is recommend is not really pertinent in this issue.

dalegaard, 2023-02-07:
Added in #2c7b578, released as v3.2.0.

closed 2023-02-07 by dalegaard