Annex-Engineering/klipper_estimator #13: Can't handle prints with Cura's `cool_lift_head`

richfelker, 2022-03-29:
If Cura's `cool_lift_head` (lift head when minimum layer time can't be met within minimum speed constraint) feature is enabled, the estimator treats each layer as a distinct print (due to Z motion back down after the lift). I'm not sure if the same happens with normal Z hop, but there probably should be some way to get the estimator to treat the pattern "increase in Z, no extrusion, return to original Z" as a hop (and account for time of hops) rather than as starting a new print.

richfelker, 2022-03-29:
Here is a sample: [mintime_test_bad.gcode.txt](https://github.com/Annex-Engineering/klipper_estimator/files/8372948/mintime_test_bad.gcode.txt)

Yes, it inserts `G4`, but I actually have an open bug against CuraEngine that it shouldn't do the dwell when the time spent lifting the head is already greater than the time it wanted to dwell; without this change, the feature is unusably slow. I think "no extrusion" is better logic for "this is not a new sequence", especially since you probably want to tolerate normal Z-hop too.

Personally I'd be okay wth the "single sequence" option since I don't do any timing of multi-sequences anyway. But I think an automatic solution would be nicer here since Z-hop is a fairly widely used feature (even if lift head for cooling isn't) and knowing the precise time spent on "hops" would be useful.

richfelker, 2022-03-29:
Ah, would it make sense for dwells just not to cause a new sequence anyway, as they're a sort of "travel" rather than "print" move?

dalegaard, 2022-03-30:
Released in 2.0.0. The test file looks to give correct output, and also, wow, that's a lot of time used just waiting :sweat_smile: 

Best regards,
Lasse

closed 2022-03-30 by dalegaard