Annex-Engineering/klipper_estimator #14: Use config-file when printer is off

viljamip, 2022-03-31:
I use Moonraker power control to turn off my printer with a relay whenever I'm not printing but I leave the Raspberry Pi always on. That causes Klipper to go to emergency shutdown as it can't communicate with the MCUs and thus Moonraker API to fetch the config is not working.

What I'd like is to have the option to use both --config_moonraker_url and --config_file parameters and then the moonraker config would be used if it is available and the config file is a fallback if moonraker is not responding. Even better would be to have a new flag which would dump the config from moonraker to the config file if moonraker is available so the offline copy is kept up-to-date.

I already tried adding the dump-config command before the post-process command in my slicer so the config file would be updated but it failed as the slicer added gcode file as a parameter which prevented the dump-config from running. If this parameter would be ignored, then this would be viable option as well.

viljamip, 2022-03-31:
For now I was able to hack this to work by adding this to my SuperSlicer post-processing script:
`./klipper_estimator dump-config > config.json #
./klipper_estimator --config_file config.json post-process`

It comments out the path for the gcode file the slicer wants to append there 😀

viljamip, 2022-03-31:
Interestingly the dump-config command does work even if the printer is turned off. Trying to use the post-process command with --config_moonraker_url results in:
`Failed to load printer configuration: Eof at 1:1 at ""`

I guess if the post-process command was able to handle fetching the config similar to the dump-config command, this would not be an issue in the first place.

dalegaard, 2022-05-26:
Hi @viljamip 

I think this would be a better fit for a script wrapping `klipper_estimator`, than directly in the tool. If you end up making such a script, feel free to submit it as a pull request and I will happily merge it :-) It would go in the "contrib" folder.

Best regards,
Lasse

closed 2022-05-26 by dalegaard