Annex-Engineering/klipper_estimator #15: Klipper Estimator does weird post-processing things with Linear Advance in Cura

psiberfunk, 2022-04-07:
As the title says, I get a weird diff when I compare the pre-processed and the post-processed files with klipper estimator when using in conjunction with FieldOfView's Linear Advance plugin.  In particular, it seems like Klipper estimator is moving around the order of the linear advance command in an odd fashion according to this diff at the beginning of the file.  Is there a reason why?

[image]

psiberfunk, 2022-04-07:
Note that this behavior is "less weird" when first generating a file without klipper estimator being used via the "inline" cura post-processing interface... When this is done instead  (produce GCODE, then post-process via the command line), the file looks different, and only the times are changed instead, and only a line added at the very end noting it was post-processed.

This command-line behavior is what I expected.. but it seems that the Cura-helper compatibility script may get executed in such a way as to create the odd behavior above .. which is a bit odd..

dalegaard, 2022-04-08:
Hi @psiberfunk 

There's a very real possibility that the Cura script is bugged. I don't use Cura myself, and wrote the script over a few hours to help. It may have weird interactions with other plugins.

Sadly I'm not really able to debug this much myself, so I hope someone else is willing to take a look at it. From your second comment it sounds like the bug is definitely in the post processing script and not `klipper_estimator` itself.

Best regards,
Lasse

dalegaard, 2022-08-05:
Hi @psiberfunk 

I'll close this for now. If someone has time to look in to the issue later on, we can re-open :)

Best regards,
Lasse

closed 2022-08-05 by dalegaard