Annex-Engineering/klipper_estimator #17: Time estimate off by 25%

jrobbom5, 2022-04-16:
I've got a model that I am slicing in SuperSlicer. When I send the gocde to the printer, the slicer says it will take 10 hours and 55 minutes, however the print completed in 7:09. I assumed that something was wrong and suspected that it was not reading the config data from moonraker, so I ran the estimator from the command line (ubuntu 21:10), and it gave exactly the same result (ie 10:55) but no errors were thrown, so I assume that means that there were no problems getting the config data from moonraker. 

I'm assuming that I have done something wrong rather than this being a bug, but I'm not sure what else I can check, so I would appreciate any tips. I can post the gcode file and superslicer profiles if that is of any help

Cheers, John

jrobbom5, 2022-04-16:
[](url
[Finger Nails - Nail_2_350.zip](https://github.com/Annex-Engineering/klipper_estimator/files/8500737/Finger.Nails.-.Nail_2_350.zip)
)

jrobbom5, 2022-04-18:
[Finger Nails.zip](https://github.com/Annex-Engineering/klipper_estimator/files/8505233/Finger.Nails.zip)
I ended up having to print this in Cura, the difference wasn't as big as I was seeing in SuperSlicer, but still well outside the 1 min per 12 hours listed in the README. 

Cura Estimate: 8:21
Klipper_Estimator: 10:20 (not including heat up times)
Actual Time: 10:01 (including ~4 mins heat up)

I hope that this gives you enough info to look into it further, but if you require anything else, please let me know. And thanks very much for making this utility available.

dalegaard, 2022-04-18:
Hi @jrobbom5 

Can you upload a config file as well? Without it, it's pretty much impossible for me to reproduce the issue.

Ideally both Klipper and estimator config, but definitely at least estimator config. You can use the `dump-config` command documented in the README.

Since both Cura and Superslicer are out by this much, I reckon it's probably something config related.

Best regards
Lasse

jrobbom5, 2022-04-18:
Hi Lasse,

Thanks very much for the reply. Sorry, I should have thought to include the config file. Here it is:
[config.zip](https://github.com/Annex-Engineering/klipper_estimator/files/8505761/config.zip)

Cheers, John

dalegaard, 2022-04-18:
Thanks @jrobbom5, I'll look into the issue asap. It make take a little while, as I'm pretty swamped with other engagements currently. Sorry about that!

Best regards,
Lasse

dalegaard, 2022-04-19:
Hi @jrobbom5 

I was waiting for a print and decided to look at this quickly. Firmware retraction speeds weren't being honored. When re-running with the config and gcode you provided, I now get 9h55m7s down from 10h20m23s. This seems to be in line with the number you provided of 10h01m(with ~4 mins heatup).

I've uploaded a version 2.0.3, could you please test it out when you get a chance? :-)

Best regards,
Lasse

jrobbom5, 2022-04-19:
Hi Lasse,

Wow! Thanks so much for that, I will test it out shortly and let you know :)

Cheers, John

dalegaard, 2022-05-06:
Hi @jrobbom5 

Did you get a chance to test this?

Best regards,
Lasse

dalegaard, 2022-05-26:
Hi @jrobbom5 

I'll close this for now. If the problem persists, please ping me here or on Discord and we'll re-open :-)

Best regards,
Lasse

closed 2022-05-26 by dalegaard