Annex-Engineering/klipper_estimator #23: Setup instructions for MacOs

LazaroFilm, 2022-07-23:
I'm on MacOs (intel) and I'm really confused on how to use this. 
The compiled osx file comes as a word document, not an executable. 

Below is the error when I try to post-process in SuperSlicer.
```
Post-processing script /Users/victorlazaro/Klipper/klipper_estimator_osx --config_moonraker_url http://192.168.68.118 post-process on file /var/folders/_l/pjyt202j6xn5l7jx__8v7kl80000gn/T/.39849.gcode.pp failed.
Error code: 126
Output:
zsh:1: permission denied: /Users/victorlazaro/Klipper/klipper_estimator_osx
```

dalegaard, 2022-07-23:
Hi @LazaroFilm 

On OS X I believe you'll need to give the downloaded file executable permissions. I don't know how to do this, as I have no experience with this operating system.

The file should definitely not be a Word document though, maybe it just appears that way because of file associations?

Best regards,
Lasse

d-nx, 2022-08-04:
I too got this same error and the problem turned out to be that the file didn't have executable permisions. 

Someone on the Klipper Discord recommended the following command (via terminal)

`chmod +x <filename>`

In your case: `chmod +x /Users/victorlazaro/Klipper/klipper_estimator_osx`

This fixed the problem for me.

closed 2022-08-05 by dalegaard