Annex-Engineering/klipper_estimator #31: Time estimate way above actual time

WolfeCub, 2022-11-01:
I wanted to test out klipper_estimator so I downloaded a few already printed gcode files and ran them through the estimate function. They all estimated longer than they actually took and some were a large amount longer. For example:

I first ran `klipper_estimator--config_moonraker_url $IP_ADDRESS dump-config > config.json` which produced the following file:

```json
{
  "max_velocity": 300.0,
  "max_acceleration": 3000.0,
  "max_accel_to_decel": 1500.0,
  "square_corner_velocity": 5.0,
  "instant_corner_velocity": 1.0,
  "move_checkers": [
    {
      "axis_limiter": {
        "axis": [
          0.0,
          0.0,
          1.0
        ],
        "max_velocity": 5.0,
        "max_accel": 100.0
      }
    },
    {
      "extruder_limiter": {
        "max_velocity": 79.82432411074329,
        "max_accel": 798.2432411074329
      }
    }
  ]
}
```

I then ran `klipper_estimator --config_file ./config.json estimate file.gcode` which produced the following output: 

```
Sequences:
 Run 0:
  Total moves:                 1694398
  Total distance:              2257241.611mm
  Total extrude distance:      28255264.679mm
  Minimal time:                1d8h23m41.648s (116621.648s)
  Total print move time:       1d6h21m21.750s (109281.750s)
  Total extrude-only time:     48m39.994s (2919.994s)
  Total travel time:           1h13m39.654s (4419.654s)
  Average flow:                582.755 mm³/s
  Maximum flow:                229638.109 mm³/s
  Average flow (output only):  621.896 mm³/s
  Phases:
   Acceleration:               11h47m18.438s
   Cruise:                     8h34m2.780s
   Deceleration:               12h2m20.180s
  Move kind distribution:
   13h6m0.391s      Internal infill
   6h9m28.506s      External perimeter
   5h45m1.344s      Internal perimeter
   2h59m47.195s     Solid infill
   2h25m27.437s     Support material
   1h10.255s        Support material interface
   25m16.625s       Top solid infill
   17m18.856s       Overhang perimeter
   14m49.282s       Internal bridge infill
   19.400s          Bridge infill
   2.108s           Custom
  Layer time distribution:
    0.000:     0.076s    10.600:  3m19.389s    21.040:    38.062s    28.240:    31.382s    37.440:    23.301s    47.750:    12.100s    54.950:     7.361s    62.750:     8.264s    72.200:  3m10.999s
    0.200: 30m32.523s    10.640:    53.701s    21.200:  3m11.667s    28.350:    31.103s    37.600:  2m59.335s    47.800:   3m9.284s    55.000:   3m7.015s    62.800:  3m16.415s    72.400:  3m16.809s
    0.240:   1m2.231s    10.800:  3m19.681s    21.240:    35.574s    28.400:  3m11.635s    37.640:    23.939s    47.840:    13.485s    55.040:     8.685s    62.840:     8.973s    72.600:  3m12.059s
    0.400: 17m18.260s    10.840:    51.164s    21.400:   3m4.793s    28.440:    30.695s    37.800:   3m3.030s    47.950:    12.168s    55.150:     7.513s    62.950:     7.139s    72.680:     3.245s
    0.440:   1m4.467s    11.000:  3m18.593s    21.440:    37.577s    28.550:    30.387s    37.840:    23.692s    48.000:   3m7.414s    55.200:   3m4.258s    63.000:   3m8.729s    72.800:   3m8.036s
[... 82 lines cut ...]
    8.800:  3m27.499s    19.240:    40.403s    27.040:    33.250s    35.640:    25.081s    46.200:   3m8.407s    53.750:     8.773s    61.550:     9.961s    70.040:     7.527s    87.600:  2m58.242s
    8.840:    54.253s    19.400:  3m11.277s    27.150:    34.076s    35.800:   3m1.062s    46.240:    15.653s    53.800:   3m1.745s    61.600:  2m56.178s    70.200:  3m10.319s    87.800:   3m6.799s
    9.000:  3m28.547s    19.440:    42.571s    27.200:  3m13.345s    35.840:    25.321s    46.400:  3m13.026s    53.840:     9.122s    61.640:    10.388s    70.240:     7.217s    88.000:   3m1.808s
    9.040:    54.275s    19.600:   3m9.510s    27.240:    32.552s    36.000:  2m59.546s    46.440:    15.527s    53.950:     8.464s    61.750:     9.875s    70.400:  3m13.451s    88.200:  2m55.993s
    9.200:  3m22.017s    19.640:    40.060s    27.350:    31.481s    36.040:    25.405s    46.600:  3m20.489s    54.000:   3m4.280s    61.800:  2m53.438s    70.440:     5.507s    88.400:  2m40.639s
    9.240:    53.329s    19.800:  3m11.231s    27.400:  3m12.603s    36.200:   3m7.348s    46.640:    14.871s    54.040:     9.156s    61.840:    10.542s    70.600:  3m17.584s    88.600:  2m29.794s
    9.400:  3m14.750s    19.840:    40.639s    27.440:    30.853s    36.240:    24.418s    46.800:  3m18.543s    54.150:     8.123s    61.950:    10.381s    70.640:     3.250s    88.800:  5m50.370s
    9.440:    54.275s    20.000:  2m54.576s    27.550:    30.973s    36.400:   3m8.386s    46.840:    14.883s    54.200:   3m5.101s    62.000:   3m2.384s    70.800:  3m16.944s    89.000:  6m13.211s
    9.600:   3m4.015s    20.040:    38.954s    27.600:   3m2.082s    36.440:    24.321s    47.000:  3m16.099s    54.240:     9.721s    62.040:    10.572s    70.920:     3.744s    89.200:  4m26.945s
    9.640:    53.087s    20.200:  2m54.912s    27.640:    31.078s    36.600:  3m10.166s    47.040:    14.806s    54.350:     8.138s    62.150:    10.044s    71.000:  3m14.807s    89.400:  4m24.070s
    9.800:   3m6.119s    20.240:    40.981s    27.750:    31.524s    36.640:    24.146s    47.200:   3m9.301s    54.400:  3m12.914s    62.200:   3m4.502s    71.200:   3m3.817s    89.600:  3m49.165s
    9.840:    54.255s    20.400:   3m1.946s    27.800:   3m5.364s    36.800:  3m12.015s    47.240:    12.182s    54.440:     9.453s    62.240:    10.719s    71.240:     3.702s    89.800:  3m22.028s
   10.000:   3m8.055s    20.440:    37.871s    27.840:    30.877s    36.840:    24.429s    47.350:    12.623s    54.550:     8.257s    62.350:     9.780s    71.400:   3m7.041s    90.000:   1m8.532s
   10.040:    52.085s    20.600:   3m2.719s    27.950:    32.520s    37.000:   3m9.313s    47.400:   3m7.559s    54.600:  3m17.790s    62.400:   3m9.149s    71.520:     3.253s
   10.200:   3m5.559s    20.640:    40.726s    28.000:  2m59.996s    37.040:    23.418s    47.440:    12.982s    54.640:     8.632s    62.440:    11.012s    71.600:   3m1.903s
   10.240:    53.507s    20.800:   3m5.259s    28.040:    32.152s    37.200:   3m5.607s    47.550:    12.194s    54.750:     8.051s    62.550:     8.831s    71.800:   3m1.373s
   10.400:  3m11.179s    20.840:    38.004s    28.150:    31.697s    37.240:    24.078s    47.600:   3m8.997s    54.800:   3m9.094s    62.600:  3m12.898s    72.000:   3m8.410s
   10.440:    52.541s    21.000:   3m2.880s    28.200:   3m9.292s    37.400:  2m59.183s    47.640:    12.780s    54.840:     9.471s    62.640:    10.066s    72.080:     3.605s
```

The estimate command says that the print should take `1d8h23m41.648s` however it actually only took `21h 30m 55s` to complete (superslicer predicted `19h 28m 55s`).

Apologies if I'm doing something stupid. Is there any setup or configuration I'm missing or is this not working properly for some other reason?

Thanks!

dalegaard, 2022-11-02:
Hi @WolfeCub 

Your process looks correct. Can you attach a gcode file for testing?

Generally discrepancies like this are caused by macro usage. The tool does not emulate macros.

Best regards,
Lasse

dalegaard, 2022-11-02:
Now that I look closer at the data, I think you are probably using absolute extrusion coordinates?

The tool defaults to assuming relative coordinates, which is different from klipper itself. I need to document why this is, but the solution is to either use reaktive coordinates in your slicer or include M82 in your _slicer_ start gcode.

Best regards,
Lasse

WolfeCub, 2022-11-02:
> I think you are probably using absolute extrusion coordinates?

Aha! Good eye. I manually edited the gcode and ran it again and it was 5 minutes shy of the actual time which adding in heat up and probing I would say is dead on.

I see where this went wrong now. My M82 command is in my start macro which the tool has no way of seeing inside and I guess superslicer doesn't emit M82 even if the relative extrusion box is unchecked. Is there a flag I can use to force it to use absolute positioning or should I just have my slicer emit an extra M82 before my start macro?

Thanks!

closed 2022-11-02 by WolfeCub

dalegaard, 2022-11-02:
At this point there's no option, but you are not the first to get bit by this and you won't be the last, so I'll see what I can do potentially to make this easier going forward, if nothing else documenting it.

Best regards,
Lasse