Annex-Engineering/klipper_estimator #32: Simplyfi3D Support

wuast94, 2022-12-16:
Simplyfi3D seem not to work, dont see the correct times in Mainsail and at the end from gcode i see this line: 

; Processed by klipper_estimator v3.0.0, no slicer detected

dalegaard, 2022-12-16:
Hi @wuast94 

As Simplify3D is paid software I don't have access to gcode files generated by it. If you slice something(any model will do, benchy, annex cube, etc) with all features related to time estimates enabled, and upload it here, I can add support.

Best regards,
Lasse

wuast94, 2022-12-16:
No Problem :)

[3DBenchy.zip](https://github.com/Annex-Engineering/klipper_estimator/files/10249708/3DBenchy.zip)

dalegaard, 2022-12-17:
Hi @wuast94 

Thank you for the sample, Simplify3D is now supported in the 3.1.0 release.

Best regards,
Lasse

closed 2022-12-17 by dalegaard

wuast94, 2022-12-18:
Would it be possible to insert the "M117 Time Left 0h6m56s" commands? S3D dont write anything in the gcode itself

And thaanks for your awesome work :)