Annex-Engineering/klipper_estimator #34: FR: Timelapse Evaluation

sangdrax8, 2022-12-30:
I notice that if I estimate a file with and without the "TIMELAPSE_TAKE_FRAME" macro after every layer, the estimates are the same.  Would it be possible to estimate the time spent doing this.  I know there are some variables in klipper which configure how fast to move, and how long to wait.  It would be really cool to be able to see how long the time lapse is adding to a print, and create even more accurate time estimates for the full print.

If I am incorrect and this is accounted for, I apologize.  I just installed this and it is amazingly better than what I was getting for estimates before.  Thanks for your great work!

dalegaard, 2022-12-30:
Hi,

This is somewhat supported via the `ESTIMATOR_ADD_TIME` directive. See https://github.com/Annex-Engineering/klipper_estimator#accurately-estimating-print_startprint_end-macros .

If/when macro support is good enough to support this use case the estimates will of course be better.

Best regards,
Lasse

sangdrax8, 2023-01-08:
Would it be possible to instruct it to add this extra time after every layer?  Then I could calcualte the approximate time per picture it takes, and it would likely be very close to correct no matter how tall or short my print code is.  Otherwise, I can't even begin to estimate the time, as it is only dependent on height.

dalegaard, 2023-01-08:
Hi,

Usually it's possible to add layer change gcode in your slicer. `ESTIMATOR_ADD_TIME` could be used here 

Best regards,
Lasse

closed 2023-01-26 by dalegaard