Annex-Engineering/klipper_estimator #37: FR: Output corrected printtime to filename

exsencer, 2023-02-06:
The postprocess mode works perfectly but I usually glance at the file name that is outputted by the slicer to gauge the print times. Can there be an option added to append the print time to the file name?

dalegaard, 2023-02-06:
Hi @exsencer 

Sadly we can't do this, as the tool is given the file by the slicer, and must change the file in place. There's no way for this process to affect the name of the file the slicer gives us.

Best regards,
Lasse

luckydevil13, 2024-06-23:
@dalegaard  look like we could do it via SLIC3R_PP_OUTPUT_NAME env variable

dalegaard, 2024-06-23:
@luckydevil13 

We can't change environment variables (they get passed to us), but the variable you mentioned brought me to this passage:

```
The post-processing script may suggest a new output file name (likely based on SLIC3R_PP_OUTPUT_NAME) by saving it as a single line into a new "output name" temp file, for example to add time stamps or sequence numbers to the final G-codes.The "output name" file name is to be created by suffixing the input G-code file name with ".output_name". PrusaSlicer will read the new name of the file and handle it properly, for example, when sending it to Octoprint.
```
This I was not aware of. I'll get this implemented next time I have time!

Best regards,
Laase