Annex-Engineering/klipper_estimator #39: Logging

paulsgale, 2023-02-17:
Hi,

Is the use of Klipper_estimator in post process mode logged anywhere? I’m not sure it’s working on my setup and wanted to check a log to see what’s happening.

Thanks.

dalegaard, 2023-02-17:
Hi @paulsgale 

There's no log as such, but a line is appended to the end of the file with a comment about what slicer was detected.

Are you looking for anything else in particular, that isn't already exposed in estimate or move dump modes?

Best regards,
Lasse

dalegaard, 2023-02-17:
Hi @pgale 

SuperSlicer, and all other slicers as far as I'm aware, do not read back gcode after post processing and so can't show the output of the estimation. The slicers either need to change, or integrate the estimation directly. There's an open SuperSlicer issue on this, but I don't believe any progress has been made on it, or if there's even interest in it.

Best regards,
Lasse

pgale, 2023-02-17:
Ok many thanks for clearing that up :) yes I don't expect SS to supoort that as current updates are pretty much non-existent since mid last year :(

closed 2023-02-20 by dalegaard