Annex-Engineering/klipper_estimator #40: Integrate kilpper estimator into OctoPrint-PrintTimeGenius plugin

eyal0, 2023-03-14:
Maybe now that this estimator exists, it can be possible to integrate into the PTG plugin?

From what I read about the kilpper estimator, the only major change that would be needed is a mode where kilpper estimator can output the mapping from file position to time. That is: when Klipper is processing file byte X, Y second have elapsed since the start of the print. Output as many of these as you like. This is how the marlin estimator works.

PTG already gathers information from the printer like the speeds and feeds. Those can be provided to the estimator by PTG.

Klipper estimation is a feature that many users have requested.

Let me know if you'd be interested in working on this together.

eyal0, 2023-03-14:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/105#issuecomment-1468647683

lbibass, 2023-04-25:
@eyal0 what features does Octoprint have that you feel are not met by the current implementation of Moonraker + Fluidd/Mainsail? 

And also, PrintTimeGenius is basically made irrelevant by this program. On every single print I've had basically perfect results.

eyal0, 2023-04-26:
I don't use PrintTimeGenius nor OctoPrint but I used to.

I guess that some users on OctoPrint would like to use PTG with Klipper.

Maybe there are some features in OctoPrint that don't exist in Moonraker? Or some plugins?

dalegaard, 2023-09-06:
Hi,

This would be for the other project to decide.

Best regards,
Lasse

closed 2023-09-06 by dalegaard