Annex-Engineering/klipper_estimator #47: Use config.js if moonraker is not online

FreezeManny, 2023-09-06:
Hello,
it would be amazing, if the klipper estimator could use the config.js in the case the 3D Printer cannot be found in the network.
Should not be difficult to integrate.
Greetings

dalegaard, 2023-09-06:
Hi @FreezeManny

This is already supported, but not currently documented. I will get to documenting it eventually, but I don't currently have the time for it. You can run `klipper_estimator -h` to see all supported options.

Best regards,
Lasse

closed 2023-09-06 by dalegaard

AndyHazz, 2025-02-09:
In case anyone else ends up here like me, this works great if you can figure out the undocumented features ..

Here's what I ended up with in Orca Slicer's post process field:

```shell
~/klipper_estimator --config_moonraker_url http://mainsail.lan --config_moonraker_cache_file ~/klipper-estimator_config.json --config_moonraker_ignore_error post-process;
``` 

It seems `--config_moonraker_cache_file` only works when used alongside `--config_moonraker_ignore_error`