Annex-Engineering/klipper_estimator #49: Increasing acceleration beyond machine's max acceleration still decreases the estimate of time taken to print

hYdos, 2023-12-11:
[benchie_1m.gcode.txt](https://github.com/Annex-Engineering/klipper_estimator/files/13631143/benchie_1m.gcode.txt)
[benchie_100k.gcode.txt](https://github.com/Annex-Engineering/klipper_estimator/files/13631145/benchie_100k.gcode.txt)
[printer.cfg.json](https://github.com/Annex-Engineering/klipper_estimator/files/13631147/printer.cfg.json)

the `_1m` benchie has accels of 1 million which is higher than the printer.cfg has set yet it performs better than the `_100k` sliced benchie

dalegaard, 2023-12-11:
Hi @hYdos 

This is what would also happen in Klipper. The `max_accel` parameter does not set an upper limit on the machine acceleration, just the maximum acceleration that will be applied to any move until the setting is changed. The setting is not very intuitively named in Klipper, so you are not the first to fall for this trap :)

Best regards,
Lasse

hYdos, 2023-12-11:
> <Annex-Engineering/klipper_estimator/issue/49/issue_event/11206170074@
> github.com>

My bad thanks for the clarification