Annex-Engineering/klipper_estimator #51: Installation

pcout, 2024-03-02:
Can someone guide me in the process of installisng this and get it configured ? Using Orca Slicer.
I found it unclear. I've tried and need some step by step help, please. 
May help others too, from what I read online, who struggle to implement this.

My setup is:
. Ender 3 S1
. Raspeberry Pi 4 running klipper (KIAUH)

Thanks.

spitfire, 2024-03-04:
No idea about Orca, but here's cura plugin (with instructions): https://github.com/pedrolamas/klipper-preprocessor

pcout, 2024-03-04:
> No idea about Orca, but here's cura plugin (with instructions): https://github.com/pedrolamas/klipper-preprocessor

It's a bit different with Cura as it uses KlipperPreprocessor, and Klipper Estimator being an option of that plugin.
Thanks anyway.

Anyone can help ?

hrvat94, 2024-03-12:
i need also help with getting started with Orca Slicer, have do the readme on the pi, where runs klipper and moonraker, but doesnt work for me, Cant run klipper_estimator. A introduction would be nice.
Thanks