Annex-Engineering/klipper_estimator #53: Error on 'M73 P0.000 R46': unable to parse 0.000

hrvat94, 2024-03-12:
It works now, but i can't print in Mainsail. 
The Error is "Error on 'M73 P0.000 R46': unable to parse 0.000"

dalegaard, 2024-03-12:
Hi @hrvat94 

Where do you see this error? Klipper supports floating point `P` values for `M73`.

Best regards,
Lasse

dalegaard, 2024-03-12:
Hi,

I'm not sure what's causing this, but it's not a `klipper_estimator` bug. Upstream Klipper supports setting floats for this field, and I've verified this on my local machine where the line `M73 P0.000 R46` runs with no issue. My guess would be it's some macro or custom module. I don't think this is the right place to debug the issue though.

Best regards,
Lasse

hrvat94, 2024-03-12:
maybe because i comment out rename_existing: SET_PRINT_STATS_INFO_BASE in the Mainsail.cfg? Its because the Ender 5 plus klipper maybe

hrvat94, 2024-03-12:
have install the original Firmware on my Ender 5 plus. Now it seems to work. The modified Firmware was the issue. Thanks