Annex-Engineering/klipper_estimator #61: Estimated print time off by a lot/seemingly random amounts

stevereno30, 2024-04-20:
For the past month, my klipper estimator has gone from being spot on to being terrible at estimating print times. In the following screenshot, the highlighted print job is when the issue started, but I have no idea what I changed between these two prints because there was a month between the two prints.
[image]

The top print in that screenshot is this gcode:
[Bolt.txt](https://github.com/Annex-Engineering/klipper_estimator/files/15048533/Bolt.txt)

Running klipper-estimator with that same gcode gives this output:
> Sequences:
>  Run 0:
>   Total moves:                 721688
>   Total distance:              1159881.251mm
>   Total extrude distance:      37097.778mm
>   Minimal time:                11h2m21.840s (39741.840s)
>   Total print move time:       8h11m24.714s (29484.714s)
>   Total extrude-only time:     31m55.008s (1915.008s)
>   Total travel time:           2h19m1.867s (8341.867s)
>   Average speed:               29.185 mm/s
>   Top speed:                   98.308 mm/s
>   Average flow:                2.245 mm³/s
>   Maximum flow:                7.021 mm³/s
>   Average flow (output only):  3.026 mm³/s
>   Phases:
>    Acceleration:               29m52.724s
>    Cruise:                     10h2m39.437s
>    Deceleration:               29m49.428s
>   Move kind distribution:
>    4h49m11.762s     Internal infill
>    3h3m27.484s      Solid infill
>    1h59m22.900s     Perimeter
>    54m6.476s        External perimeter
>    6m36.632s        Bridge infill
>    5m19.307s        Top solid infill
>    3m51.093s        Overhang perimeter
>    23.781s          Skirt/Brim
>    2.155s           Custom

If needed, this is my printer config:
[printer.txt](https://github.com/Annex-Engineering/klipper_estimator/files/15048566/printer.txt)

stevereno30, 2024-04-20:
This issue has something to do with Klipper's [Skew Correction](https://www.klipper3d.org/Skew_Correction.html) feature. Commenting out my [skew_correction] section gives much more accurate estimates (off by a couple minutes instead of 10+ hours.

stevereno30, 2024-04-23:
Skew wasn't the only issue.  There is something else, as my prints continue to be off by hours in some cases.
[image]

stevereno30, 2024-04-23:
Something about the post-process...um, process via PrusaSlicer is borked. The following prints were the same plate in PrusaSlicer exported twice. The 3.5 hour one was exported without doing anything special. The 1.5 hour one came about from the following procedure: 

1. The post-process script was removed from PrusaSlicer
2. The file was exported without post processing to my desktop
3. The file was processed via a terminal with the estimate command, resulting in a print time of 1h37min
4. Post process script removal was reverted using PrusaSlicer's "undo" button
5. File was sliced, and the resulting file has a print time of 1h37min

[3h39m.gcode.txt](https://github.com/Annex-Engineering/klipper_estimator/files/15083837/3h39m.gcode.txt)
[1h37m.gcode.txt](https://github.com/Annex-Engineering/klipper_estimator/files/15083839/1h37m.gcode.txt)

fjpanag, 2024-04-24:
I have the same issue.

I do not use skew correction, but I do use Prusa Slicer v2.7.4, on Fedora 39.

The attached G-code is predicted in Prusa Slicer with a duration of 3:42.  
Klipper estimator predicts 7:57!

The actual print time was 3:03.

[image]

[wrong_estimation.gcode.txt](https://github.com/Annex-Engineering/klipper_estimator/files/15100517/wrong_estimation.gcode.txt)

stevereno30, 2024-04-27:
Here is more evidence for this bug. I sliced a gcode, exported to my desktop without any post-processing, then ran klipper_estimator's estimate command via terminal two times in a row without changing anything. This is what my terminal output was: 
`PS C:\Users\steve> C:\Users\steve\Documents\KlipperProcesses\klipper_estimator.exe --config_moonraker_url http://192.168.0.10 estimate C:\Users\steve\Desktop\TEST.gcode
Sequences:
 Run 0:
  Total moves:                 992220
  Total distance:              1469218.344mm
  Total extrude distance:      37280.289mm
  Minimal time:                18h27m15.998s (66435.998s)
  Total print move time:       13h18m8.225s (47888.225s)
  Total extrude-only time:     49m29.742s (2969.742s)
  Total travel time:           4h19m37.781s (15577.781s)
  Average speed:               22.115 mm/s
  Top speed:                   82.578 mm/s
  Average flow:                1.350 mm³/s
  Maximum flow:                6.044 mm³/s
  Average flow (output only):  1.872 mm³/s
  Phases:
   Acceleration:               48m46.022s
   Cruise:                     16h50m10.048s
   Deceleration:               48m19.677s
  Move kind distribution:
   7h9m21.124s      Perimeter
   3h54m3.079s      Solid infill
   3h25m23.485s     External perimeter
   3h4m35.952s      Internal infill
   28m10.698s       Bridge infill
   23m48.365s       Top solid infill
   1m28.997s        Overhang perimeter
   22.214s          Skirt/Brim
   1.834s           Custom
  Layer time distribution:
    0.000:     0.049s     8.000: 14m30.979s    16.000:   3m6.773s    24.000:    13.248s    32.000:    15.263s
    0.200: 25m18.298s     8.200: 11m40.325s    16.200:  3m27.954s    24.200:    14.691s    32.200:    12.944s
    0.400: 22m21.810s     8.400: 11m41.782s    16.400:  3m26.003s    24.400:    16.372s    32.400:    11.801s
    0.600: 24m35.248s     8.600: 11m42.087s    16.600:  3m25.318s    24.600:    16.002s    32.600:    12.108s
    0.800:  25m1.912s     8.800: 11m40.714s    16.800:  3m23.952s    24.800:    15.923s    32.800:    12.013s
    1.000: 23m40.045s     9.000: 11m35.998s    17.000:  3m15.901s    25.000:    16.731s    33.000:    11.657s
    1.200: 15m12.620s     9.200: 11m14.980s    17.200:  1m17.564s    25.200:    12.960s    33.200:    11.625s
    1.400: 15m52.684s     9.400: 11m40.590s    17.400:    50.693s    25.400:    14.280s    33.400:    12.102s
    1.600: 15m21.708s     9.600: 11m30.437s    17.600:    49.300s    25.600:    14.644s    33.600:    12.226s
    1.800: 14m59.991s     9.800: 11m14.806s    17.800:    47.403s    25.800:    14.649s    33.800:    12.479s
    2.000: 14m43.862s    10.000: 10m54.057s    18.000:    47.077s    26.000:    14.658s    34.000:    12.092s
    2.200:  15m0.602s    10.200: 10m53.562s    18.200:    48.128s    26.200:    14.662s    34.200:    11.892s
    2.400: 16m22.838s    10.400: 10m12.418s    18.400:    46.735s    26.400:    14.669s    34.400:    11.959s
    2.600: 16m39.977s    10.600: 10m24.621s    18.600:    46.132s    26.600:    14.711s    34.600:    11.698s
    2.800: 16m15.863s    10.800: 10m23.749s    18.800:    46.429s    26.800:    14.694s    34.800:    11.698s
    3.000: 16m21.927s    11.000:  10m8.750s    19.000:    42.039s    27.000:    14.696s    35.000:    11.744s
    3.200: 18m27.667s    11.200:  8m42.638s    19.200:    40.674s    27.200:    14.704s    35.200:    13.741s
    3.400: 19m29.126s    11.400:  8m21.421s    19.400:    39.757s    27.400:    14.739s    35.400:    14.310s
    3.600:  19m1.436s    11.600:   8m2.932s    19.600:    36.542s    27.600:    15.555s    35.600:    14.029s
    3.800: 18m54.573s    11.800:  7m53.088s    19.800:    35.454s    27.800:    16.487s    35.800:    15.615s
    4.000:  19m2.510s    12.000:  7m58.817s    20.000:    32.593s    28.000:    16.506s    36.000:    17.552s
    4.200: 16m59.638s    12.200:   6m3.942s    20.200:    24.650s    28.200:    19.510s    36.200:    14.722s
    4.400: 16m27.838s    12.400:  5m46.412s    20.400:    21.524s    28.400:    17.182s    36.400:    15.079s
    4.600: 15m58.091s    12.600:  5m44.195s    20.600:    21.276s    28.600:    17.303s    36.600:    15.082s
    4.800:  16m3.644s    12.800:  5m45.990s    20.800:    21.375s    28.800:    17.442s    36.800:    14.281s
    5.000: 15m45.590s    13.000:  5m44.491s    21.000:    21.560s    29.000:    16.270s    37.000:    15.031s
    5.200: 15m36.268s    13.200:  5m41.520s    21.200:    23.635s    29.200:    14.987s    37.200:    15.117s
    5.400: 15m31.678s    13.400:  5m32.090s    21.400:    24.137s    29.400:    14.788s    37.400:    14.993s
    5.600: 15m31.232s    13.600:  5m32.478s    21.600:    24.211s    29.600:    14.914s    37.600:    14.653s
    5.800:  15m6.131s    13.800:  5m35.243s    21.800:    24.956s    29.800:    14.977s    37.800:    14.517s
    6.000: 14m41.879s    14.000:  5m42.057s    22.000:    25.188s    30.000:    14.861s    38.000:    14.483s
    6.200: 15m40.457s    14.200:  6m24.277s    22.200:    23.968s    30.200:    14.196s    38.200:    14.500s
    6.400: 16m31.777s    14.400:   6m8.706s    22.400:    23.573s    30.400:    14.219s    38.400:    14.505s
    6.600: 15m53.601s    14.600:  6m10.465s    22.600:    23.525s    30.600:    13.947s    38.600:    15.743s
    6.800: 15m35.413s    14.800:  6m13.968s    22.800:    23.438s    30.800:    14.393s    38.800:    16.064s
    7.000: 15m58.501s    15.000:  6m16.146s    23.000:    24.012s    31.000:    14.169s    39.000:    16.604s
[... cut ...]
    7.000:  6m40.363s    15.000:  2m32.444s    23.000:     9.914s    31.000:     6.210s    39.000:     6.785s
    7.200:  6m30.897s    15.200:  1m26.211s    23.200:     7.133s    31.200:     6.221s    39.200:     6.735s
    7.400:  6m18.743s    15.400:  1m19.127s    23.400:     5.411s    31.400:     6.242s    39.400:     6.664s
    7.600:   6m9.706s    15.600:  1m16.945s    23.600:     5.449s    31.600:     6.172s    39.600:     0.735s
    7.800:   6m3.446s    15.800:  1m16.600s    23.800:     5.595s    31.800:     6.620s`

b-fitzpatrick, 2024-04-27:
I updated to v3.7.2 from v3.4.0 to deal with the minimum_cruise_ratio change. I saw the same as @stevereno30: different estimates for the same input. I had similar results with v3.7.1, but v3.7.0 seems to be giving me consistent results (well, five times in a row for the same input).

dalegaard, 2024-04-27:
Hi @stevereno30 and @b-fitzpatrick 

Sorry, I've been swamped lately and have not had a chance to look at this until now.

I've just released version 3.7.3 which should fix this and #62. If you could test it out and report back, that would be a big help. Thanks in advance.

Again sorry for the delay!

Best regards,
Lasse

b-fitzpatrick, 2024-04-27:
@dalegaard, I'm getting consistent and seemingly-reasonable results with v3.7.3. Thank you!

stevereno30, 2024-04-28:
No need to apologize. I can't be impatient when this is free software!

My last print estimate, post-processed with 3.7.3, was off by 3 seconds compared to actual print time. Data point of one, but that seems about as perfect as it can be!

closed 2024-04-28 by dalegaard