Annex-Engineering/klipper_estimator #62: override minimum_cruise_ratio some times

vit-sh, 2024-04-23:
Test configuration file with script in the test.zip archive. The script cyclically launches klipper_estimate with a dump into a configuration file with the launch number. The result is in the resilt.zip archive. The default value for the max_accel_to_decel parameter is periodically substituted.
[test.zip](https://github.com/Annex-Engineering/klipper_estimator/files/15079176/test.zip)
[result.zip](https://github.com/Annex-Engineering/klipper_estimator/files/15079175/result.zip)

dalegaard, 2024-04-28:
Hi @vit-sh 

This should be fixed now. Can you confirm on latest release? 

Best regards, 
Lasse

vit-sh, 2024-04-28:
I confirm. Everything works perfectly.

closed 2024-04-28 by vit-sh