Annex-Engineering/klipper_estimator #64: post-process doesn't seem to work.

blakeyjason, 2024-06-03:
Hi,

 I'm working on getting my Ender3 V3 KE to show progress on prints that I've uploaded using Fluidd/Moonraker.  The job prints fine, but the progress never updates.

 Looking at the code that Creality's slicer generates, I see that it embeds

;TIME_ELAPSED:d.dddddd

values.

I've updated my Prusa Slicer to add 

;TIME_ELAPSED:1.234567

on layer change, but post-process never changes my default 1.234567 to actual time values.   What am I missing?  The estimator runs fine from PrusaSlicer, but the generated output file has the same TIME_ELAPSED values as before.

Once I get this up and running, I'd be happy to contribute a how-to on PrusaSlicer + klipper_estimator + Ender3 V3 KE.

Thanks!

dalegaard, 2024-06-03:
Hi @blakeyjason 

Check the last line of the file, it will say `Processed by ...` and tell you which slicer was detected.

Best regards,
Lasse

dalegaard, 2024-06-03:
Hi @blakeyjason 

As of today there is no way to force a specific slicer version. While it would be possible to update the `TIME_ELAPSED` code to work for other slicers, I'm a bit weary of that because of potential differences in how the various time formats would work. The slicer detection code is there to ensure that we do sane things for all slicers instead of risking outputting bogus values.

I was under the impression Creality OS was just Klipper though. What stack are you using here? As far as I'm aware, PrusaSlicer does not natively output `TIME_ELAPSED` lines?

Best regards,
Lasse

blakeyjason, 2024-06-03:
The problem is not with Creality OS - that is pretty much klipper.   Side-loaded Fluidd displays the updates fine.  But the native Creality display on the printer expects to see TIME_ELAPSED values from the gcode (that's what I'm seeing in output from the Creality Slicer, called "Creality Print").

My stack is PrusaSlicer -> Estimator -> MoonRaker -> Fluidd -> The default Creality display app running on the printer.

dalegaard, 2024-06-03:
Okay, now I understand.

This isn't supported right now, and I don't have much time to implement it sadly. I would like to implement a way to do replacements of some template format in the gcode with time estimates. This would also work for #63 .

Basically the idea would be that instead of putting `;TIME_ELAPSED:1.234567` in your slicer, you'd put something ala:
```
;TIME_ELAPSED:{estimator#current_time_seconds}`
```
And estimator would replace it with the right value.

I will keep this open until a mechanism like that is implemented.

Best regards,
Lasse

blakeyjason, 2024-06-03:
That would be great!  Nice and general purpose.  

Thanks for thinking about it for me.