Annex-Engineering/klipper_estimator #66: Support for the acceleration change set by orcaslicer with Emit Limits in G-Code option

sb53systems, 2024-07-10:
How to include the acceleration change set by orcaslicer with Emit Limits in G-Code option ?
[image]
[image]

Thank you.

sb53systems, 2024-07-14:
this small time difference is due to this missing parameter  
  
[image]

dalegaard, 2024-07-28:
Hi,

I'm not sure what's being reported here. 

Best regards, 
Lasse

sb53systems, 2024-07-28:
Hey Lasse,

I think that Klipper_Estimator_Script uses the maximum printer acceleration
provided in the Config.json file to estimate the time of each movement and
its flow rate average, and does not support the frequency defined in the
G-Code, Example :

SET_VELOCITY_LIMIT ACCEL=1400 ACCEL_TO_DECEL=700
G1 F9000
G1 X35 Y48
SET_VELOCITY_LIMIT ACCEL=3000 ACCEL_TO_DECEL=1500
G1 F3600
G1 X40 Y50

I noticed this because the time estimated by Klipper_Estimator is always
the same whether the option  (Emit Limits to G-code) is activated or not.
[image: image.png]

My Config.json File is below

{
  "max_velocity": 300.0,
  "max_acceleration": 6000.0,
  "minimum_cruise_ratio": 0.5,
  "square_corner_velocity": 5.0,
  "instant_corner_velocity": 1.0,
  "mm_per_arc_segment": 0.1,
  "move_checkers": [
    {
      "axis_limiter": {
        "axis": [
          0.0,
          0.0,
          1.0
        ],
        "max_velocity": 5.0,
        "max_accel": 200.0
      }
    },
    {
      "extruder_limiter": {
        "max_velocity": 79.82432411074329,
        "max_accel": 1596.486482214866
      }
    }
  ]
}

I thank you in advance for your support

Best Regards..
Salim.

sb53systems, 2024-08-26:
@dalegaard any future support for this issue ?  
Your help would be very valuable !  
  
Best regards,
Salim

dalegaard, 2024-08-27:
Hi @sb53systems 

`set_velocity_limit` is supported, so I'm not sure what's wrong here. I don't use `klipper_estimator` anymore and I don't have time to debug the issue. Sorry.

Best regards,
Lasse

closed 2024-08-27 by sb53systems

sb53systems, 2024-09-30:
@dalegaard , I confirm that the time estimated by the current version of Klipper Estimator is really accurate (+/- 1S)  
the only difference between the estimated time and the one that the Klipper Firmware calculates, is the purge line (included in the START_PRINT marco) which takes 10s in my case  
[image]  
  
Thank you very much for implementing this tool, it helped me a lot in my last project.  
  
# G-Code Flow and Temperature Controller (using Klipper_Estimator V 3.7.3)
https://github.com/sb53systems/G-Code-Flow-Temperature-Controller  
  
Best Regards.  
Salim