Annex-Engineering/klipper_estimator #69: Support for inserting M73

sesse, 2024-09-25:
Hi,

I'm using klipper_estimator via the klipper_postprocessor script in Cura, and the estimates are pretty much amazing. However, Klipper cannot really use the klipper_estimator estimates for percentage completed, for some reason; it can only show a static countdown. I see that Fluidd can use M73 commands as input for its estimations, and I see some code in klipper_estimator that can replace M73 commands; but Cura doesn't insert any, so there's nothing to replace. Is there a way to have klipper_estimator insert M73 on its own? (I know there's some _other_ Cura post-processing script that can insert M73 commands, but it also seems to insert M117 messages that I don't want, and I can't really get it to work all that well in general.)

dalegaard, 2024-10-27:
Hi @sesse 

At this point there's no such support. It could be added, but I no longer use `klipper_estimator` myself and don't have the time to add support.

I'll leave this issue open so the feature request doesn't get lost.

Best regards,
Lasse

nathanielhudson, 2024-10-29:
@sesse for what it's worth, I had the same problem and solved it by just inserting "blank" M73 commands (as in, just an M73 with no arguments) to my custom layer start gcode in my slicer. klipper-estimator then replaces those with M73s with valid args and it's all good.

sesse, 2024-10-29:
@nathanielhudson It's a nice hack, but unfortunately, klipper_estimator only intercepts M73 for certain G-code dialects, and Cura is not one of them. (Not to mention that Cura doesn't support inserting arbitrary commands at layer start; this is something that can be fixed with another modification script, though.)